mic159/octoprint-grbl-plugin #2: cannot detect archive format

The3rdIcon, 2017-06-14:
After upgrading pip, I'm still getting the error below.

> Cannot unpack file /tmp/pip-TBiaVW-unpack/octoprint-grbl-plugin (downloaded from /tmp/pip-gVEwVa-build, content-type: text/html; charset=utf-8); cannot detect archive format
Cannot determine archive format of /tmp/pip-gVEwVa-buil

Is this an issue with the Pi for the Plug in?

mic159, 2017-07-21:
Hmm, weird. It works for me:

```
$ pip install https://pypi.python.org/packages/26/7d/85dea16ac65ef3bd97f6b7c78865c7bfe245f3a91fc14ae1e71e7d749abf/octoprint-grbl-plugin-1.0.1.tar.gz --no-cache-dir
Collecting https://pypi.python.org/packages/26/7d/85dea16ac65ef3bd97f6b7c78865c7bfe245f3a91fc14ae1e71e7d749abf/octoprint-grbl-plugin-1.0.1.tar.gz
  Downloading octoprint-grbl-plugin-1.0.1.tar.gz
Installing collected packages: octoprint-grbl-plugin
  Running setup.py install for octoprint-grbl-plugin ... done
Successfully installed octoprint-grbl-plugin-1.0.1
```

How did you try to install it?

scottmil, 2017-08-13:
According to the developer of Octoprint, you need to use the version of pip that comes with Octoprint to install plugins.

mic159, 2017-09-05:
I have updated the link to the tarball in the plugin repo, can you try again now?

closed 2017-09-13 by mic159