mic159/octoprint-grbl-plugin #3: Stop after errors

nebnes, 2017-07-21:
The process stops after the following errors in the terminal. I use grbl 0.9 (the cnc works fine with the grbl sender "Rabbit Mill")

`Recv: error: Expected command letter`

`Recv: error: Invalid gcode ID:33
WARNING! Received an error from the printer's firmware, ignoring that as configured but you might want to investigate what happened here! Error:  Invalid gcode ID:33\x0d\x0a`

`Recv: error: Invalid gcode ID:24
WARNING! Received an error from the printer's firmware, ignoring that as configured but you might want to investigate what happened here! Error:  Invalid gcode ID:24\x0d\x0a`

mic159, 2017-07-23:
Do you know what GCODE commands were sent to GRBL at the time?
Could you turn on the "Log communication to serial.log" function in OctoPrint and share what gcode was sent just before the errors, and how far into the gcode it was?

Sounds like some invalid GCODE was received, see the table on this page: https://github.com/grbl/grbl/wiki/Interfacing-with-Grbl
(search for `Invalid gcode ID`)

I'm also wondering if it would be possible to upgrade to GRBL 1.1 as well?

nebnes, 2017-07-24:
The log before the errors : 
```
Send: G1 X212.1051 Y150.5262
Recv: ok
Send: G4 P0
Recv: ok
Send: M03 S180
Recv: ok
Send: G4 P1
Recv: ok
Send: G1 F3000.000000
Recv: ok
Send: G2 X206.412 Y122.7055 I-70.8228 J-0.
Recv: ok
Send: G2 X190.061 Y98.9199 I-67.2871 J28.7422
Recv: ok
Send: G2 X165.7521 Y83.1329 I-52.3227 J53.9578
Recv: ok
Send: G2 X136.842 Y77.5438 I-28.9101 J71.9764
Recv: ok
Send: G2 X83.2828 Y99.2477 I0. J76.9368
Recv: ok
Send: ?$G
Send: G2 X61.5789 Y150.5262 I49.7247 J51.2786
Recv: <Run,MPos:291.469,151.088,0.000,WPos:211.469,141.088,0.000>
Send: G2 X83.2828 Y201.8048 I71.4285 J0.
Recv: ok
Send: G2 X136.842 Y223.5086 I53.5593 J-55.233
Recv: error: Invalid gcode ID:24
WARNING! Received an error from the printer's firmware, ignoring that as configured but you might want to investigate what happened here! Error:  Invalid gcode ID:24\x0d\x0a
Recv: error: Bad number format
WARNING! Received an error from the printer's firmware, ignoring that as configured but you might want to investigate what happened here! Error:  Bad number format\x0d\x0a
Recv: error: Invalid gcode ID:24
WARNING! Received an error from the printer's firmware, ignoring that as configured but you might want to investigate what happened here! Error:  Invalid gcode ID:24\x0d\x0a
Changing monitoring state from 'Printing' to 'Operational'
Send: G4 P0
Send: ?
Send: ?$G
Recv: ok
Recv: <Idle,MPos:270.060,108.928,0.000,WPos:190.060,98.928,0.000>
Recv: ok
```
I have isolated this command `G2 X212.1051 Y150.5262 I-65.1298 J-27.8207` and I get ` error: Invalid gcode ID:33` it's strange because we have already use this gcode with Rabbit Mill without errors. 

I use grbl 0.9 because I need a servo motor on the z-axis ( https://github.com/robottini/grbl-servo ).

mic159, 2017-09-13:
Sounds like the buffers are getting overflowed.
It's possible that might be related to the "temperature" poll?
Could you try changing the temperature poll interval to be something very large and see if its more stable?

aftersight, 2018-09-10:
I know I'm ressurecting the dead here, but I was having the same issue, and implemented this solution and it worked. I set the polling interval to 86400 seconds (once per day) and have no issues.