mic159/octoprint-grbl-plugin #6: GRBL - OctoPrint Connection Errors.

JorgenCollinson, 2017-09-06:
hey,

The firmware on the Arduino 2560 Mega works when using Easel... so i think the Firmware is working all ok.

Hoping from some advise please on using GRBL/OctoPrint as am getting UnHandled Communication Errors
I installed the GRBL Plugin.

Changed the Hello Command to "M5" as recommended in the instructions.
Removed the "Commands that always require a Checksum" to blank.

Terminal shows the following when connecting.
> Connecting to: /dev/ttyACM0
> Changing monitoring state from 'Offline' to 'Opening serial port'
> Connected to: Serial<id=0x758481f0, open=True>(port='/dev/ttyACM0', baudrate=115200, bytesize=8, parity='N', stopbits=1, timeout=10.0, xonxoff=False, rtscts=False, dsrdtr=False), starting monitor
> Changing monitoring state from 'Opening serial port' to 'Connecting'
> Send: N1 M5*39
> Recv: 
> Recv: Grbl 0.9j ['$' for help]
> Changing monitoring state from 'Connecting' to 'Operational'
> Send: N2 M115*36
> Recv: error: Unsupported command
> Changing monitoring state from 'Operational' to 'Error:  Unsupported command\x0d\x0a'

hoping you can offer some advice.
thanks in advance.

mic159, 2017-09-07:
Can you try turning off line numbers (the N1, N2, etc)?

JorgenCollinson, 2017-09-08:
hey,
thanks for the feedback... 
Question 1 : Any idea how i do that in OctoPrint, am sure i just need a different handshaking command ?

Question 2 : the Hello Command in the instructions says to put "M5" as the GCode hello command... this command is what i expect to use from the GRBL firmware to turn my Spindle on...

thanks in advance.

techpad-marcb, 2017-09-13:
Trying this out as well and this is what i get after connection.

`Connecting to: /dev/ttyUSB0
Changing monitoring state from 'Offline' to 'Opening serial port'
Connected to: Serial<id=0xaed6a270, open=True>(port='/dev/ttyUSB0', baudrate=115200, bytesize=8, parity='N', stopbits=1, timeout=10.0, xonxoff=False, rtscts=False, dsrdtr=False), starting monitor
Changing monitoring state from 'Opening serial port' to 'Connecting'
Send: M5
Connection closed, closing down monitor
Recv: ok
Changing monitoring state from 'Connecting' to 'Operational'
Send: ?$G
Send: M21
Recv: <Idle|MPos:0.000,0.000,0.000|FS:0,0|WCO:-86.748,21.418,-36.138>
Recv: [GC:G0 G54 G17 G21 G90 G94 M5 M9 T0 F0 S0]
Recv: ok
Recv: error:20
Changing monitoring state from 'Operational' to 'Error: 20\x0d\x0a'`

to turn off the n1 commands you need to go under "Features -> Enable automatic firmware detection" and there is an option at the bottom to turn it off.

Any ideas on the error i'm getting when connecting using grbl 1.1f

mic159, 2017-09-13:
M5 is to turn the spindle off in GRBL.
See https://www.shapeoko.com/wiki/index.php/G-Code#Machine_Options_.28M.29

You can use any gcode in the hello that will generate an "ok" message, as that what OctoPrint needs.

That error you are getting there (`error:20`) is about an unknown GCODE.
`M21` is not a valid GRBL gcode, and that is what is causing it to abort there.

Find where the M21 is coming from and turn it off. The only hint from googleing it is that it might be related to SD cards.

electr0sheep, 2018-02-16:
I had some issues connecting to a LinkSprite Mini CNC using the latest version of OctoPrint (1.3.6)
To connect successfully I did the following:

1. Serial Connection > Advanced > "Hello" Command = M5
2. Serial Connection > Advanced > Commands that always require a checksum = _blank_
3. Features > Enable SD support = _uncheck_
4. Features > Enable automatic firmware detection = _uncheck_
5. Features > Send a checksum with the command = Never (only becomes visible after step 4)

Now my problem is that I can move the cnc beyond it's bounds, resulting in some serious gear grindage. Don't think that's an issue with this plugin, but I'm pretty new to all this so who knows.

mic159, 2018-02-21:
@electr0sheep Does your LinkSprite Mini have end stops?
(those are physical switches at the end of each axis that tell the motor to stop/has gone too far?)

I would have to double check, but I dont think OctoPrint has smarts around stopping you from manually jogging the head off the edges, or trimming your gcode.

On my laser engraver, I use end stops to prevent any mistakes from causing damage, it also uses them to reset 0,0 to the same spot every time.

synman, 2019-03-15:
I'm guessing development on this plug-in is dead.  I'm trying to decide if modifying it and doing a pull request will be reviewed / accepted or if I should just use it as a reference for a plug-in of my own.

mic159, 2019-03-15:
@synman I can review pull requests and merge them.
Access to a machine to test with is a little tricky right now for me, but if your willing to test it, then I don't have any issues shipping it.

asps946701, 2019-03-21:
Hellew  @mic159 
How to delete or modify M115 ? When I using Octoprint to connect my grbl ,   it run to error:20 .
Thank you very much.
--------------------
Send: M5
Recv: 
Recv: Grbl 1.1g ['$' for help]
Changing monitoring state from "Connecting" to "Operational"
Send: M115
Recv: error:20
Changing monitoring state from "Operational" to "Error: 20"
Changing monitoring state from "Error: 20" to "Offline (Error: 20)"
Connection closed, closing down monitor
----------------------