mic159/octoprint-grbl-plugin #11: Abandoned?

synman, 2022-12-27:
I'm going to recommend this plugin be marked as abandoned and repointed to "Better Grbl Support" in the plugin repository.  Credit / authorship will be carried forward as this plugin sparked a ton of development for the Better Grbl Support plugin.

mic159, 2022-12-28:
I no longer use Octoprint, and I am happy for you to take over with your new plugin.
Your plugin has a tonne of new features and config options, which is awesome, nice work!

The only thing that throws up question marks is that on config save, there is a code path that sends HTTP requests. Seems very strange.

synman, 2022-12-28:
Welcome to the next generation of Grbl!  (http://wiki.fluidnc.com/en/config/overview).  What you are seeing is the config file upload process for machines running FluidNC.

https://github.com/synman/Octoprint-Bettergrblsupport/blob/devel/octoprint_bettergrblsupport/__init__.py#L540

[image]

FluidNC (like its predecessor Grbl_Esp32) runs on a 32 bit Espressif Systems 2 core SoC with Wi-Fi.  In addition to running Grbl, it also hosts a WebUI that provides a number of embedded functions, such as firmware OTA updates, SDCard support, and, in the case of FluidNC, hosting its primary machine configuration file.

[image]

Pretty cool eh?